WelterRocks/WESPOTA #4: CAN-bus

wolfgangr, 2019-01-16:
I'd just add this as a stub here.

the CAN-bus protocol is used not only in mobile context, but also in many home automation environments, like heating systems, electrical installations...

So sniffing your heater's CAN bus to read temperatures, control commands, OK?

Basically, CAN does not more than send and receive some packets of bits.
So it could easily be mapped to UDP or MQTT, I think.
Maybe the software side could be implemented along SendSerial-command of tasmota.

I have postponed my CAN related stuff - mainly due to a lack of time for the moment.
I have some gadgets for a testbed around, 
- MCP2502
- MCP 2115
- Tiny-CAN II USB-module for PC
- AT90 CAN32
- ...???
So if anybody wants to drop in here, please let me know.

btw: 
this is a screenshot from testing on raspi.
... just networking ....

```
pi@wjr-raspi-helloCAN:~/CanBerry $ ip link 
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:9f:7c:e8 brd ff:ff:ff:ff:ff:ff
3: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/can 
```

VerboteneZone, 2019-01-18:
I love the CAN bus idea, because I have some other projects running on CAN bus, like this one: **https://github.com/VerboteneZone/libbmwxcan**
which is a controller library for BMW E and partially F series. I have tested this already using a NodeMCU, a CAN-Bus Module (doesnt have the part no right now) and was able to read cars telemetry and control things like windows and door locks.

CAN-Bus is a really wide field. I like to support it, but this will grow rapidly...

wolfgangr, 2019-01-18:
I just think of retrofitting my combine and my potato harvester.

The last combine I sold in really good used condition was 35 yrs old. Its still operating, now, 5 yrs later.
Maybe there was a transistor in the flasher.... and a couple in the alternator.
But definitely no MCU.

The one I currently own still has a cabinet of relays at the side. But was the first one with GPS, touch panel control and such modern stuff when it was developped 20 yrs ago.
When you ask for service today, they send you the salesman :-\
So I could end up with a 15 to pile of scrap in the one and a close to 200 k€ bill in the other hand....
Or contracting the harvesting out, to have mor time to play with IoT gadgets :-)))

Same stuff with fork lifts.
Have three of them to be scrapepd for dysfunctioning control systems.
Repair is more expensive than get a 'new' old one.
But neither lasts for long nor can be considered as sustainable, I think.

VerboteneZone, 2019-01-20:
Not really a problem. The module, I am currently using to remote start my car is an I2C based CAN-Bus interface, which could be "easily" supported by tasmota/wespota. But therefore, we need software defined buttons (some kind of virtual outputs, without hardcoded maximums) and a rule store, where to save and deploy the needed can messages. Therefore the 512 byte rule store is to small, to handle some more comlex rulesets. What do you think?

wolfgangr, 2019-01-23:
@VerboteneZone 
Oliver, do you plan to invest work on this issue in the foreseeable future?
If not, I'd suppose to close it for the moment.

wolfgangr, 2019-01-24:
### postponed ####

closed 2019-01-24 by wolfgangr