WelterRocks/WESPOTA #5: WiFi distance sensing

wolfgangr, 2019-01-16:
In theory, you can calculate distance by measuring the distance from your WiFi-gadget to one (maybe also more) base stations.

Just by accident, I recieved a note that some project along this thought was revived:
https://github.com/cnlohr/esp8266rawpackets/issues/8

If it works out,  I think it easily can be added as "sensor" into the tasmota/WESPOTA framework

VerboteneZone, 2019-01-18:
Maybe some kind of triangulation, too? Therefore a connection is not needed. You only need the APs MACs and the RSSI levels. This will not be very accurate, but surely it will be a nice gadget to have.

wolfgangr, 2019-01-18:
Trinangulation - yes, of course. 
LPS aka "local positioning system".... forklift in a store, just to mention....
RSSI - well, depends on the application.
But maybe just a one-night-project, if we ha a high-level scripting language.

For ToF-Measurement I'm afraid that you need to hack the AP as well.
But that's not an issue, if this can be a <10€ gadget.

But I have to keep my todo-list manageable.
Life is coming into the way ... customers threatening with orders.... :-O
put this to 'wait and see' for the moment.

wolfgangr, 2019-01-24:
# postponed #
looks like there are major obstacles to get precise timing info
no easy solution visible
closed for now since not on top priority

closed 2019-01-24 by wolfgangr