WelterRocks/WESPOTA #12: 16-Kanal Relais Modul Relay MODBUS RTU RS485

wolfgangr, 2019-01-17:
Having one of those in the pipeline from Hong Kong:
12V 16-Kanal Relais Modul Relay MODBUS RTU RS485 PLC Controller Serial Schalter
http://www.ebay.de/itm/283155629937
~ 20 €

looks like the well known 4-Channel modules, just quadruple the size :-)
(unfortunately I stil cannot post images here)

I quote from the description:
> 3. 6 Commands: "Open", "close", "Momentary", "Self-locking", "Interlock" and "Delay".
> 4. MODBUS RTU command control mode, the maximum delay is 255 seconds.
> 5. MODBUS commands can be made serial HyperTerminal (serial assistant) or "Modbus Poll" enter.
> 6. Under the MODBUS command mode, this relay module can support up to 64 devices in parallel. 

This raises two additional question that I will put into separate issues:
- shall we manage **modbus** as a separate level of abstraction?
- 64 devices * 16 relays yield 128 switched lines -does this fit into the current hardcoded handling in tasmota?

VerboteneZone, 2019-01-18:
I think the question is, what will the vendors do in the near future. And MODBUS is the way, if you want to control many outputs. So MODBUS will be a mus have. As I told, we will drop the hardcoded maximum stuff, so that things like MODBUS could be supported.

VerboteneZone, 2019-01-18:
Nice board. I will order one for testing.

wolfgangr, 2019-01-18:
brother in soul, eh ;-)))