WelterRocks/WESPOTA #13: Modbus and Modbus TCP

wolfgangr, 2019-01-17:
First time these days I encountered the topic "Modbus" when I was ordering these 16 channel relay modules: https://github.com/VerboteneZone/WESPOTA/issues/12

Second encounter was when I yesterday coined the issue on frequency converters: https://github.com/VerboteneZone/WESPOTA/issues/7
I was looking for the seria protocol for the Danfoss-converters, but it looks like you have buy additional hardware modules to do so. Many of them on the list carry a "modbus" label.

Find a concise description here: https://en.wikipedia.org/wiki/Modbus
> Modbus has become a de facto standard communication protocol and is now a commonly available means of connecting industrial electronic devices.[

.....
> Example of frame in hexadecimal: 01 04 02 FF FF B8 80 

Doesn't seem that more complicated than what we send to the 4-Channel-LCTech modules.

> Versions of the Modbus protocol exist for serial port and for Ethernet and other protocols that support the Internet protocol suite. 

Doesn't that sound like that what IoT is about?
WESPOTA thingies might be both master and slave, using TCP, UDP, serial and whatever else the standard defines.

Personally, at the moment, I don't have need for modbus beyond the 16-ch relais.
But other people may have - now or in the future.
So at least, I would keep this issue in mind when the architecture is refurbished, so that it easily can be integrated at a later time.

VerboteneZone, 2019-01-18:
Thats exactly the WESPOTA idea, to support as many hardware, as possible. But first we need to find a new way for the driver and board management. As I said before, I like to have vendor, main- and daughterboard selectors, which will also have an influence to the needed or available drivers.

wolfgangr, 2019-01-18:
What do you think of my forth idea?
If we could implement any higher level driver stuff in a higl level language, that might ease out things a lot.
So I'd give this a preference, if any other decisions on architecture depend on it.
Or do you have another alternative - one that might be easier at hand?

VerboteneZone, 2019-01-20:
I like the idea, yes, I really do. But first of all, I have to look at forth by myself, because of my missing experience with that. I will have a look, as soon as possible.

wolfgangr, 2019-01-23:
Maybe someone has done it already?
Just accindentially encountered this link.
I'll keep it here for pickup later on.

https://www.esp8266.com/viewtopic.php?t=1916&start=4
https://www.esp8266.com/viewtopic.php?p=14818#p14818

>Re: MODBUS Server (TCP) ... #14818
>By radim100 - Fri Apr 17, 2015 8:54 am
>Hi , This is not exactly what you need but may be good starting point .
> Modbus TCP bridge . TCP in MODBUS RTU on RXD and TXD out .