WelterRocks/WESPOTA #14: Handling of arbitrary large numbers of inputs / outputs

wolfgangr, 2019-01-17:
As far as I could figure out, the maximum number of relais (.. switches, buttons, analog ins, temperature sensors....) is somewhat hardcoded in tasmota.

I alredy postet 16 ch relais modules, connected max 64 of them to a single bus.
There are I/O expanders on the 12c which offer siḿiliar magnitudes of multi I/O.

The oneWire bus for the DS18B20 allows many .... (neary "unlimited") number of devices connected to one single bus.

I think code refurbishment shoud aim at keeping the nuts and bolts of the number within the driver.
Towards the rest, this should be transparent.
Well, on the other hand, it might sometimes be relevant to refer to a specific item in some kind of a device tree - compared to pci or usb tree in linux.
And of course, this label should not change, when devices are added or removed.
Just think of the udev quagmire.... 

Hm..... some thinking required, I'm afraid....

VerboteneZone, 2019-01-18:
Yes, i totally agree. The idea also is, to set the useable channel number at runtime and remove the hardcoded stuff. In conjunction with advanced rule processing, I would like to have software defined buttons in webui.

The way, drivers are loaded in tasmota is a pain, but i think they did it because of the dev environment. The first thing I thought about is to throw away and no longer support the Arduino ide, which makes things easier. But also PIO has only limited support, if you plan to load things dynamically. There are some complicated ways, right now. But there will be a solution.

wolfgangr, 2019-01-18:
Agree about PIO.
Nice Edit and browse tool, but heavy layer of obfuscation / opaqueness.
Had maybe some 5 crashes till now, one X-server freeze (on a debian - hadn't such athing for years)
In the end it costs more than it safes, in my opinion.
I'd actually prefer a makefile solution.
Not even the automagic one offered with Arduino-ESP, but one as simple as possible, to provide full control.
oops., should that go to a new issue?

VerboteneZone, 2019-01-20:
The PIO magic is done with python so far. People would like it, because of ATOM and PIO is a quick and handy solution, if you like 'Klicki-Bunti' desktops. For me, a debian bash shell, python, pip, the python-pio stuff and a simple joe editor will also do the job, without the overhead, crashing the X-servers (and also win-boxes). So, the automatic nightly builds of WESPOTA at https://wespota.verbotene.zone/nightly/ do work this way. And yes, maybe this should go to a new issue ;-) maybe...