WelterRocks/WESPOTA #16: No releases so far?

heikoh81, 2019-01-17:
Hi there,

my LC Technology 4 relay board has just arrived by mail this week.
I also have a LC Technology 2 relay board.
Both come with a ESP-01S (1M).
I tried ESPEasy (works for 2-relay board).

I would give WESPOTA a try, but couldn't find any releases.
Where could I find a release that has native support for my LC Technology 4 relay board?

Best regards,
Heiko (Stuttgart, Germany)

wolfgangr, 2019-01-18:
well ... sad to say ... not yet .
Pull the current trunk and hope the best.
I think the only difference to tasmota main stream are the changes I made to get the LCTech 4CH running.
Double check the configuration settings, because I tweaked a lot of them and tried them only with my personal user_config_secret.h
I don't even know if Oliver had the chance to test it himself, before he created this fork.

But if you try it - succesful or not - please leave a report here.

'=============================

@VerboteneZone Oliver created this fork about one week ago
As far as I can figure out, from restricted internet access in some hospital :-O

So first of all: Oliver, I wish you the very best regarding your health :-)

Next I dropped all my memory in issues here.
So, yes, still much honour to be earned :)

VerboteneZone, 2019-01-18:
Thank you very much. I am currently not at home right now.

The main differences to TASMOTA are the LC Tech implementation, a certificate store implementation (is used later for some security improvements and the support of WPA2 Enterprise). Also the directory scheme has changed to support the upcoming new drivers, also there are some minor changes. I will describe them later. The nightly builds, with default config can be found at https://verbotene.zone, but there is no guarantee, that they will work. 

My wife brought some SONOFF basic devices and a 2 channel LC Tech board. I will test WESPOTA later today, but first I have to go to therapy. Thanks for supporting :-)

wolfgangr, 2019-01-18:
@VerboteneZone 
Oh, I had neither diffed nor scrutinized your changes, Nor do I know all your ideas in mind.

@heikoh81 
This sentence
> I think the only difference to tasmota main stream are the changes I made to get the LCTech 4CH running.

refers to this tag on my own git
https://github.com/wolfgangr/Sonoff-Tasmota/tree/LCT_noblock

maybe if you just want a quick hack to get your 4CH-relays running, but otherwise stick to tasmota as close as possible, this might be preferrable for you.

WESMOTA is going t be a greater rework endeavour, as I expect it.

VerboteneZone, 2019-01-18:
Oh, I just forget to say: For custom configurations, which are not pushed to git, use src/include/config/wespota_alternate_config_sample.h and copy it wespota_alternate_config.h in the same folder.

Use -DHAVE_ALTERNATE_CONFIG in platformio.ini or at command line, to use wespota_alternate_config.h.

heikoh81, 2019-01-18:
Thank you all for your answers and Gute Besserung to Oliver.

> maybe if you just want a quick hack to get your 4CH-relays running, but otherwise stick to tasmota as close as possible, this might be preferrable for you.

Yes, I just want to use my 2-/4-channel LC-Boards with Tasmota-like firmware (as I like the MQTT implementation of Tasmota).

So, do I just have to download & flash this release to my ESP-01S?
https://github.com/wolfgangr/Sonoff-Tasmota/releases/tag/LCT_noblock

heikoh81, 2019-01-18:
I just noticed there is no .bin inside LCT_noblock.zip.
Any chance for a precompiled .bin that I can flash with ESPEasy?

heikoh81, 2019-01-19:
Could someone add or send me a working .bin for LCTech 2ch/4ch boards?
I tried original tasmota with custom rule as described in wiki, not working.

As far as I understand, there is a working solution out there, but not precompiled :-(

VerboteneZone, 2019-01-20:
@heikoh81 you could try https://wespota.verbotene.zone/nightly/
There are precompiled binaries with LC-Tech enabled configurations

You just have to select LC-Tech Relay 1, 2 or 4 channel. You can Flash this binary directly to an ESP-01 device. Also, be Aware.... no warrenty at all. 

I have flashed this firmware to all my sonoff and lc-tech devices yesterday and everything seems to work, until now. Just the 4 channel lc-tech board is untested, yet. Maybe wolfgangr can give it a try.

heikoh81, 2019-01-20:
Thank you very much, flashing with ESPEasy was no problem.

> I have flashed this firmware to all my sonoff and lc-tech devices yesterday and everything seems to work, until now. Just the 4 channel lc-tech board is untested, yet. Maybe wolfgangr can give it a try.

I can confirm that this is working like charm on LCTech 2 relay board.

However, after putting the exact same ESP-01S to my LCTech 4 relay board and selecting LCTec 4 channel, there is no reaction to Toggling at all.
Also, if I select 2 channel or 1 channel while ESP-01S is placed in 4 channel board, no reaction.

So for the 4 channel Version some things need to be fixed.
How can I help?

Greetings, Heiko

heikoh81, 2019-01-20:
Update!

Luckily, I have ordered 2 LCTech 4ch boards. One with 5V, one with 12V.
I first tried the 5V board --> no reaction on the relays.
Now I put the same ESP-01S into the 12V version ==> SUCCESS, all 4 relays switching.

So Wespota is working perfectly.

What could be the reaon for the 5V-Version not to work?
While on the 12V version there is a red LED constantly on and a green one blinking, on the other there is a blue LED on.

heikoh81, 2019-01-20:
UPDATE2:
After pressing BTN01 on the 12V version, the blue LED is now on also on my 12V board, and the green LED is flashing. Relays are no longer working!
The blue LED is on after restart and even after POWER cycle.
How can I turn the blueLED off again? I tried pushing BTN01 (and BTN02) several times, also holding it down for a few seconds.

The blue LED is always on and the greenLED is flashing.

heikoh81, 2019-01-20:
UPDATE3:
Holding down Btn01 while powering up changed Mode back to Mode 1.
Blue LED is Hot Spot mode as far as I found out from various product description pages.

wolfgangr, 2019-01-23:
Hi @VerboteneZone and @heikoh81 ,
This sounds like the issue is solved.
So way me close this down, just to clean up a little bit around here?

heikoh81, 2019-01-23:
Yes, you can close the issue.

wolfgangr, 2019-01-24:
solved - closed

closed 2019-01-24 by wolfgangr